reversebias/mitosis-hardware #4: A regulation and capasitors?

biberforce, 2017-06-11:
There is a regulator in schematics(U2). What's that name? and you need a filter capacitor for Vın and Vout.
https://electronics.stackexchange.com/questions/67859/how-to-select-capacitor-values

TyberiusPrime, 2017-06-14:
The bom lists a '1117 3.3v regulator in SOT223 (many types available)'

I second the addition of the caps. The output can't be stable without a Vout capacitor. 
This might be mitigated by caps on the rf board but it should really be as close to the 1117 as possible.

AMS1117-3.3 datasheet suggests 22uF tantal.

reversebias, 2017-06-14:
Yeah, I was relying on the pro micro for input caps, and the RF board for output caps. For the ceramics that are on the RF board, it's actually an advantage that they're further away, the increase in ESR is necessary for stability. The value is lower than the 10uF recommended, but for such a low output current (30mA peak on the RF module), the main pass transistor is in a much higher impedance state, which pushes out the dominant pole. I agree that best practice would be to include caps anyway, but it's been measured to be completely stable, and I was going for low part count. I'll probably change the regulator if I do a major revision, I just had these on hand when I was building.

closed 2017-11-06 by reversebias