reversebias/mitosis-hardware #5: Extend pads under rf module

TyberiusPrime, 2017-06-11:
Don't know if it's my clone of board being milled a bit small but I'm about a tenth of a mm short which makes it a pain to solder one side (actually haven't managed it yet...).

Think it would be much easier if the landing pads were extended half a millimeter inwards giving a bit of tolerance.

TyberiusPrime, 2017-06-14:
Or maybe aim for this module in the next revision?

http://www.waveshare.com/product/Core51822.htm

reversebias, 2017-06-14:
There have been others with the same problem, seems like there are undocumented versions of the module with different sizes. This is on my list to fix when I have time. 

As for that module, I considered it, but it was significantly harder to get a mirrorable design with a through hole footprint.