reversebias/mitosis-hardware #6: Make the lassercut files avaliable in a free format

Aventura5, 2017-07-17:
The 3dm files can't be opened by any non-paid software and the laser cutting services I'm aware of don't take raster, only vector files so the PNG is not useful. SVG is kind of a lingua franca, everything can import it but DXF is also good.

kdb424, 2018-08-18:
Both parts are PCB, so they will need to be sent to someone like dirtyPCB. You won't be able to use other materials as both halves serve an active role in the electronics. This can be seen in the build log here. https://imgur.com/a/mwTFj

joric, 2018-11-14:
I converted 3dm to dxf here: [mitosis-laser.dxf](https://raw.githubusercontent.com/joric/mitosis-hardware/joric_kicad/cnc/mitosis-laser.dxf) (reddit thread)[https://www.reddit.com/r/MechanicalKeyboards/comments/92b3xw/mitosis_bottom_layer_lasercutting_files/]