reversebias/mitosis-hardware #12: LiPo for keyboard?

xkortex, 2019-01-14:
Very happy with my Mitosis so far with one small exception: I left it on my shelf for ~2 months and the battery was dead. I thought it should get more like 6 in standby. The variant I got from Flashquark has a housing which needs to be removed to get at the battery; this is a tad inconvenient. 

I'm looking at putting in a lipo so I can at least change it without taking the housing apart. Is 3.7V something the board can handle natively, or would I need to put in a 3 or 3.3v regulator as well? The nRF nRF51822 specifies `Supply voltage range 1.8 V to 3.6 V` which feels like pushing my luck.

datagrok, 2019-01-14:
@xkortex I have the same kit, and [I have done exactly that](https://github.com/reversebias/mitosis/issues/1#issuecomment-344452103). I don't know if I'm stressing any limits of the ICs or something but I've been using 3.7v LiPos plugged into the +3.3v and Ground programming header pins, with no additional regulator circuitry, and it seems to work great!

xkortex, 2019-01-16:
Oh hey, @datagrok I'm using your keyboard as a basis! Did you ever get that I2C hack splice working?

datagrok, 2019-01-16:
@xkortex I did! But I only got as far as proof-of-concept before life happened and I had to set the whole thing aside for... months? (I actually had to go read up in my repo to remember what you were even talking about!) :sweat_smile:

Anyway here's a little video of my Mitosis receiver with a $0.50 speaker and $2.50 I2C OLED display attached, running some demo code. It's hard to see but there's a wire in there routing the green LED previously on pin 3 to the Pro Micro pin 4, so I get to use I2C and all the LEDs still work.

[image]

Hopefully I'll get a chance to tidy it up, test, and submit a new PR upstream in the next several weeks.

xkortex, 2019-01-18:
Very cool stuff :) :+1:

xkortex, 2019-02-23:
@datagrok Have you ever had issues with the blue and red LED channels not working simultaneously? For reference, I'm using the [Flashquark version](https://flashquark.com/product/gb-mitosis-wireless-split-ergonomic-keyboard-w-acrylic-case/). On your layout, I can get the led to light up red or blue, but when I press both, it just goes red. Smells like undervoltage to me, but my header pins are all clipped short so I can't easily go snooping (and don't have the time to be messing much with hardware until summer). Also I cannot get green to show, with or without  `MITOSIS_DATAGROK_I2CHACK`

Also I found out the hard way that the space between the bottom and the acrylic is juuust to thin for Adafruit's thinnest LiPo to squeeze in without scraping a sharp through-hole wire....and sparking...and leaking electrolyte. Oops :P

datagrok, 2019-02-26:
@xkortex sorry it took me a couple days to notice your question. I got my kit from a FlashQuark group buy too, with the acrylic case (which I have since decided i dislike, and removed.)

I haven't had issues like you describe; each component of my RGB LED works fine regardless of which other components are on or off. I think undervoltage is unlikely since the receiver is USB powered. I've plugged a whole bunch of LEDs into the unused pins and had them all lit at once without issues. Not to mention the I2C OLED display.

I suspect that you're somehow ending up using code intended for my modified hardware where I have rerouted the green LED pin. Maybe the MITOSIS_DATAGROK_I2CHACK toggle mechanism isn't working properly for stock pin assignment or something. I haven't looked at this in a while; let me study it a bit and reply again later.

> I found out the hard way that the space between the bottom and the acrylic is juuust to thin for Adafruit's thinnest LiPo to squeeze in without scraping a sharp through-hole wire....and sparking...and leaking electrolyte. Oops :P

[image]

datagrok, 2019-03-13:
@xkortex I finally got a chance to sit and think about this a bit more.

I checked that my code wasn't the culprit, then I re-read your description:

> - Green is totally MIA.
> - Red/blue can't light at the same time
> - Red/blue should be totally unaffected by portD, BUT if the bit 0b10 (annotated SDA) is set low ( i.e. setbits(PORTD, ~(0b00000010), 0b00100010); ) it'll nerf the RGB LED.

I think you're on the right track to suspect undervoltage. Your last bullet point makes me think there is a short to VCC from PD1/SDA (pro micro pin 2). So when you set that pin low, which would normally light the green LED, current takes the less-resistance path from VCC instead so neither the green nor any other LEDs light. (Referring to [the schematic](https://github.com/reversebias/mitosis-hardware/blob/master/schematics/receiver.pdf) should help here)

But since you're not getting a total brownout, which i think would reset the chip, I suspect that the current is still flowing through the resistors on that line. I'd suspect a bad solder join under the RGB LED. I remember it being pretty hard to solder without getting a solder blob trapped underneath it and bridging some pins.

If that's not it, find a magnifying glass and check the joins on those tiny resistors. Or use a multimeter to verify the amount of resistance through each line is correct, and that there's no continuity between lines.

Anyway, I can confirm my LEDs are working and exhibiting no problems when lit simultaneously. So it's probably a solder issue, not a problem with the circuit design. Best of luck troubleshooting from here!

datagrok, 2019-03-13:
@joric thanks for the note! I've been watching your bluetosis firmware project with interest, and I love that you were able to Mitosis-ify the Jian design to take advantage of the 100mm x 100mm pcb prototype discounts and the pcb-plate cutouts trick. Fantastic!

chris-ryu, 2022-04-29:
> 

@datagrok If you are the one who made the datagrok keymap please help me.
I know this issue thread is not the right place to ask for help, but I see that your id is datagrok and ask for help.
I love datagrok's red, blue and purple layering system. :)
A few days ago the receiver module on my mitosis keyboard broke.
So I am trying to apply this bluetosis firmware made by @jorian.
I am trying to apply datagrok design to bluetosis firmware.
I'm trying to start from @joric's jorian branch, but I don't know how to implement something like MO among the QMK macros used in datagrok.

Any advice or help would be appreciated.